thomaslepoix/Qucs-RFlayout #1: Build issues with QtCreator MinGW

bvernoux, 2019-10-19:
I have encountered multiple compilation issue with actual trunk when using QtCreator with MinGW64 (tested with QtCreator 4.10.1 & Qt 5.12.4 on Windows7 SP1 64bits)

1. First main issue is a conflict with some reserved define
https://github.com/thomaslepoix/Qucs-RFlayout/blob/master/microstrip/element.h#L24
```
#define _X		0
#define _Y		1
```
Those define return a compiler error as they are reserved keywords ... 
 I have fixed that by doing a refactor(in whole code) on those two constants renamed to EL_X & EL_Y I'm pretty sure we could find a better name ...

2. Other issues to be fixed are related to string escape sequence like "\.net" to be fixed to "\\.net" I do not understand how it build on your side as it is something standard to do in any C/C++ code with string.

- https://github.com/thomaslepoix/Qucs-RFlayout/blob/32db3cb410f175471e9078a9d53ae5badae5f426/parser.cpp#L82
Fixed by: ` n_net=regex_replace(n_sch, r_sch, "\\.net"); `

- https://github.com/thomaslepoix/Qucs-RFlayout/blob/32db3cb410f175471e9078a9d53ae5badae5f426/layoutwriter.cpp#L24
Fixed by: `  regex r_sch("\\.sch$");` 

- https://github.com/thomaslepoix/Qucs-RFlayout/blob/32db3cb410f175471e9078a9d53ae5badae5f426/layoutwriter.cpp#L25
Fixed by: ` regex r_basename("^.*?([^\\/]*)\\.sch$");` 

After that all build fine (I have just removed LIBS += -lglut -lGLU in the Qucs-RFlayout.pro)
I have not done lot of tests so far and I plan to check all work fine
Anyway it will be great to fix those small issues in order to rebuild the project with Qt on Window ...
Just a last word congratulations & thanks for that amazing tool !

easyw, 2019-10-24:
@bvernoux 
have you a cloned repo with the fix?
which are the requisites to build it? I'm using a Msys2 chain to build kicad...

bvernoux, 2019-10-24:
I have done other fixes I will push them in my repo if you are interested as so far the author does not answer

thomaslepoix, 2019-10-24:
Oh sorry, I didn't see your issue! Thank you, I am not a Windows user so your feedback is totally welcome. I personally use GCC or clang on Ubuntu.

About defines, I planned to make them enums. I will take a look at your fixes soon! :)

bvernoux, 2019-10-28:
Also I'm very impatient to see the OpenEMS integration as today it is very hard to use OpenEMS (even if 
 https://github.com/dlharmon/pyopenems seems very interesting)

bvernoux, 2019-11-07:
Issue fixed since https://github.com/thomaslepoix/Qucs-RFlayout/pull/2 is merged

closed 2019-11-07 by bvernoux

thomaslepoix, 2019-11-10:
Thank you! If you have some suggestions about your experience with OpenEMS and features you would find useful, I'm open :)

bvernoux, 2019-11-10:
I just dream to use OpenEMS with a GUI (like http://www.sonnetsoftware.com the Demo is amazing) as so far it is so complex to do all by script/python (see https://github.com/dlharmon/pyopenems) with some obscure parameters ...
Also an other project similar to OpenEMS https://github.com/halbux/sparselizard seems really amazing it will be great to check if it is easier to integrate versus OpenEMS and also to compare them in accuracy ... especially for EM simulation related to PCB microstrip designs...

halbux, 2021-06-10:
@bvernoux I am there if any help is needed for that. Sparselizard has been used this year to simulate a full 3D patch antenna EM wave propagation with > 30 million degrees of freedom. There is mostly everything available that any person doing EM simulations could expect (S-parameters and PMLs are missing but that's an easy task to add, it requires no code change, just to find the right math) + extra tools like hp-adaptivity that can be very convenient. Also, nonlinear EM wave propagation was tested this year as well as EM waves through lossy ferrite materials.