thomaslepoix/Qucs-RFlayout #3: Cross compilation Windows from Linux

thomaslepoix, 2020-02-16:
I'd like to provide Windows binary and to document the build process from sources.

- Arch linux provides several libraries for MinGW, like Qt, Boost, etc. that distributions like Ubuntu does not provide. So here is a container dedicated to cross-compilation for Windows with MinGW : https://hub.docker.com/r/burningdaylight/docker-mingw-qt5

```sh
~ $

    sudo docker run -v $PWD/out:/workdir/out -u root -it burningdaylight/docker-mingw-qt5 /bin/bash

        git clone https://github.com/thomaslepoix/Qucs-RFlayout
        mkdir Qucs-RFlayout/build && cd Qucs-RFlayout/build
        x86_64-w64-mingw32-cmake ..
        make
        cp ../out/* /workdir/out
        cp /usr/x86_64-w64-mingw32/bin/* /workdir/out
        cp /usr/x86_64-w64-mingw32/lib/qt/plugins/imageformats/* /workdir/out
        cp /usr/x86_64-w64-mingw32/lib/qt/plugins/iconengines/* /workdir/out
        cp /usr/x86_64-w64-mingw32/lib/qt/plugins/platforms/* /workdir/out
        exit

    ls out
```

The build looks fine but wine crashes when I try to run it in graphic mode. `wine qucsrflayout.exe -G`

- The error messages from popup & console :

```
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

```
wine: Call from 0x7bc5b50c to unimplemented function KERNEL32.dll.RaiseFailFastException, aborting
```

<br>

- I am wondering if it is a wine misconfiguration or if I am doing something wrong with Qt libraries but I have no Windows installation to check myself.

- Plus, what are the commands we have to write in a Windows shell to build it from sources?

<br>

@bvernoux may I ask you to take a look? :)

bvernoux, 2020-02-16:
I have built it on my side see archive [qucsrflayout_1.0.0_MinGW64.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/4209463/qucsrflayout_1.0.0_MinGW64.zip) 
It is built with QtCreator 4.11.0 on Windows7 Pro with Qt5.12.6 & MinGW64 (mingw730_64) and it works fine see archive containing all dependencies (maybe it is why you have a crash)

Extract of my deployment script used **windeployqt_release_Qucs-RFlayout.bat**:
```
set qt_version=5.12.6
set qt_execpath=%cd%\build-Qucs-RFlayout-Desktop_Qt_5_12_6_MinGW_64_bit-Release\release

set qt_execname=Qucs-RFlayout.exe

set qtpath=D:\Qt\%qt_version%\mingw73_64\bin\
set windeployqt_execpath="%qtpath%\windeployqt.exe"
set PATH=%qtpath%;%PATH%

%windeployqt_execpath% --release %qt_execpath%\%qt_execname%
COPY /Y "D:\Qt\%qt_version%\mingw73_64\bin\libstdc++-6.dll" "%qt_execpath%\"
Pause
```

thomaslepoix, 2020-02-17:
Thank you!

I miss some dll to run your program, plus if I try to run it with my dll from the process above, I got the same crash as with my executable. So my wine config is probably wrong.

Can you run this one [qucsrflayout-1.0.0-win64.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/4216248/qucsrflayout-1.0.0-win64.zip) ?

bvernoux, 2020-02-18:
I have tested it and it do not launch with error:
[image]
This error means you shall create **platforms** directory which contains file **qwindows.dll** 
You shall add other dependencies like in my release with dependencies solved https://github.com/thomaslepoix/Qucs-RFlayout/files/4209463/qucsrflayout_1.0.0_MinGW64.zip

Note: My version is a built with stable Qt LTS using Qt5.12.6, yours is based on Qt5.14.1.0 so that explain if you are mixing DLL from different Qt version it crash.

thomaslepoix, 2020-02-27:
Ok I got it!

I would be kind of relieved if the problem of __windeployqt__ came from recent Qt versions instead of the MinGW version.

As pointed [here](https://stackoverflow.com/questions/29678352/how-to-use-qts-windeployqt-in-linux-fedora), peldd I used as replacement can't detect dlls that are conditionally loaded at runtime such as __qwindows.dll__.

So this runs for me [qucsrflayout-1.0.0-win64.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/4258997/qucsrflayout-1.0.0-win64.zip)

bvernoux, 2020-02-27:
I confirm your latest build qucsrflayout-1.0.0-win64.zip work fine on native Windows7 SP1 64bits
Just for information when launch from icon, it open a additional command window with "GUI mode" which is not very nice

thomaslepoix, 2020-02-27:
Nice, thank you!

I cannot reproduce with wine, does it resolve it if you use `cout` instead of `log_err` [here](https://github.com/thomaslepoix/Qucs-RFlayout/blob/master/src/main.cpp#L109) ?

bvernoux, 2020-03-01:
So far I do not have time (and no space on my SSD) to play with docker/CMake as I'm using Windows Qt Creator with Qt5.12.x which build fine (I just need to modify Qucs-RFlayout.pro to remove TARGET) and the native built exe does not create/open additional command window with "GUI mode" in that case.
If you can provide a new cross compiled exe with cout instead of log_err (or even without any log) I could test.

thomaslepoix, 2020-03-02:
`cout` is enabled only with the `--vebose` flag at runtime but I think I spotted the difference between the QMake and CMake builds : The [`WIN32` flag of the CMake `add_executable()` command](https://cmake.org/cmake/help/v3.17/prop_tgt/WIN32_EXECUTABLE.html)

Yes, of course, here is the rebuild : [qucsrflayout-1.0.0-1-x86_64.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/4272810/qucsrflayout-1.0.0-1-x86_64.zip)

Although I think it is now obsolete, why the `TARGET = $$PWD/out/qucsrflayout` line is problematic for you?

bvernoux, 2020-03-06:
In fact when I build the source with `TARGET = $$PWD/out/qucsrflayout ` and Qt 5.12.7 I have this error

> Makefile.Release:146: *** multiple target patterns.  Stop.

If I build the source with  `TARGET = $$PWD/out/qucsrflayout ` and Qt 5.14.1 all is ok so it seems this feature is incompatible with Qt LTS 5.12.x and working only since Qt 5.14.x

I confirm your rebuild : qucsrflayout-1.0.0-1-x86_64.zip is ok and now Console windows appears now.

thomaslepoix, 2020-03-06:
Really nice!

Ok, good to know

Thank you very much for your time! :)

closed 2020-03-06 by thomaslepoix