thomaslepoix/Qucs-RFlayout #6: Simple Stub - Syntax Error in .m file

dom11990, 2020-07-01:
I'm trying out the openEMS features but unfortunately the .m file seems to have some syntax issues. I've attached my schematic. Any idea what is going wrong? 
I built against this commit (Date:   Tue Jun 30 04:22:48 2020 +0200): 9e65020dce55dcc3b9e80b52c0fdebde1c54cb7b

[openems.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/4860034/openems.zip)

thomaslepoix, 2020-07-02:
My bad! I fixed it!

dom11990, 2020-07-02:
Thanks for the fast reply! Unfortunately I have a different problem with the script now. It says ERROR : Unknown argument : --gui
I'm running this in Windows, perhaps it is related? The contents of arg_list is:

  [1,1] = --gui
  [2,1] = --persist
  [3,1] = --eval
  [4,1] = edit 'C:\path\to\file\openems2.m'
  [5,1] = --gui

dom11990, 2020-07-02:
I should add that those arguments were added by octave automatically, I simply clicked run after I opened the script.

thomaslepoix, 2020-07-03:
The command line is the following `octave <octave args> script.m <script args>`.

Those default octave arguments are not needed and I don't know why it adds an argument to the script. By the way `--gui` is not supported by the script, you can use `--help` to see supported arguments.

If you encounter difficulties with the octave GUI, I suggest you to run the script from a terminal.

commit 45201469 (2020-07-02): fix #6 : typo
closed 2020-07-03 by dom11990