thomaslepoix/Qucs-RFlayout #8: Mesh lines not respecting 1/3 rule

dom11990, 2020-07-02:
The meshing does not seem to respect the 1/3 rule. For one the y mesh lines seem to be off, and the x lines do 1/3 copper on one side of the MLIN and 2/3 copper on the other.

at a Via
[image]

at a Tee
[image]

thomaslepoix, 2020-07-03:
There is no way to totally automate the mesh generation, it is still to the user to ensure the mesh is correct.

So here we go! If you run the script with `--no-smoothmesh --no-process --no-postprocess`, you will see that the default mesh resolution is way too large. You can reduce it by growing its divisor `metal_div` in the script, or rerunning qucsrflayout with `--oems-metalres-div`. 200 seems to be fine.

[image]
[image]
[image]

Warnings come from the OpenEMS `SmoothMesh()` function if some manually placed lines are too close.

Note that if some components produce really close lines, you might have to remove some manually by commenting lines in the script. Component names are commented for each line so they are not so painful to find. (It is not the case here)

I will write some tutorials ASAP :)

thomaslepoix, 2020-07-03:
As I see you uses a via, do you think something about that #9 ?

dom11990, 2020-07-03:
I added my thoughts on #9 . Yes some tutorials would be very helpful :) I would love to see a more graphical approach to configuring the different options. Perhaps even configuring the mesh in the export GUI as well as other settings such as the convergence criterion, mesh density, port offsets, active ports, frequencies for the far-field calcualtion, etc. Thoughts?

thomaslepoix, 2020-07-03:
It is becoming quite stable but the OpenEMS feature is still under heavy development, so for now the GUI does not allow yet everything that the CLI does. But it should and it will.

About the script interface, it is a little more tricky as OpenEMS has its own workflow and I don't want and/or cannot distort it too much. However a kind of graphical runner could be good and feasible.

commit fbe31a29 (2020-07-20): fix #8 : add oems tutos
closed 2020-07-22 by thomaslepoix