thomaslepoix/Qucs-RFlayout #10: Qucs Studio support

dom11990, 2020-07-03:
I've been using Qucs Studio over classic Qucs lately because it has a number of features that make it a viable replacement for top dollar software (in certain use cases such as mine). It would great if we had full support for their schematic type or at least a description of what doesn't work so we can look out for it.

thomaslepoix, 2020-07-03:
QucsStudio is mostly compatible with Qucs, microstrip components are supported for both, CPW aren't for any and QucsStudio's striplines neither.

The principal limitation is that QucsStudio does not have a CLI to export a netlist from a schematic, so to convert a QucsStudio schematic you still needs Qucs to be installed.

dom11990, 2020-07-03:
I've been posting a number of questions to the QS guys. I will add this as a feature request and perhaps they will include it in the future.

dom11990, 2020-07-08:
Is this netlist sufficient for what you need?

http://qucsstudio.xobor.de/t74f4-Exporting-a-netlist-via-CLI.html#msg222

Looks something like this:

# QucsStudio 2.5.7  C:/Users/Dominik/.qucs/taps_project_prj/harm.sch

SPfile:X2 gnd _net0 "LTC5564_-10dBm.S1P" "polar" "cubic" "short" "no" "1"
Eqn:Eqn1 "mlin_w=0.441mm" "yes"
.SP:SP1 "lin" "0.1 GHz" "16 GHz" "2000" "no" "1" "2" "none"
SUBST:Subst1 "4.5" "0.254 mm" "35 µm" "0.03" "1.72e-8" "0.15 µm"
MLIN:MS1 _net1 _net2 "Subst1" "0.00397064" "0.00307842" "Hammerstad" "Kirschning" "26.85"
MLIN:MS4 gnd _net2 "Subst1" "0.00194699" "0.00812157" "Hammerstad" "Kirschning" "26.85"
MLIN:MS3 gnd _net1 "Subst1" "0.000579273" "0.00519903" "Hammerstad" "Kirschning" "26.85"
Pac:P1 _net1 gnd "1" "50 Ohm" "0 dBm" "1 GHz" "26.85"
MLIN:MS2 _net2 _net0 "Subst1" "0.000782836" "0.00647354" "Hammerstad" "Kirschning" "26.85"
Eqn:Eqn3 "s11=max(range(dB(S[1,1]),3.0e9,5.0e9)+15" "s11_close=max(range(dB(S[1,1]),3.75e9,4.25e9)+20" "yes"

thomaslepoix, 2020-07-09:
Yes it is the same format. Not satisfied with the manual procedure they gave you, however I could add an entry to specify a netlist instead of automatically call Qucs to create it.

dom11990, 2020-07-10:
Yeah the manual aspect is a bit tedious, but a think a field where you can drop the netlist directly would make sense. It is certainly a feature I would use!

thomaslepoix, 2020-07-10:
Here it is in text mode! The GUI will be reworked later :)

commit c5a7e1d1 (2020-07-10): fix #10 & some bugs
closed 2020-07-10 by thomaslepoix