thomaslepoix/Qucs-RFlayout #11: Support for variables in schematics

dom11990, 2020-07-03:
The typical flow for a filter design is to set up the architecture, be it stub or what have you, define parameters you want to tune, and let the optimizer run. For higher order filters (10+), it can be really tedious and error prone, to copy the optimized variables into a fresh schematic to be used in the extraction / generation process. If variables were supported, the workflow would be almost seamless because the same schematic could be used. 

PS- I really appreciate the work you've put into this tool. I plan to use it for a number of filters and design up to 40 GHz. I think we could use this opportunity to learn much about the user experience and technical performance. I will provide some measurements of results once the boards come in. I'm not sure how motivated you are to make it more user-friendly, or if your intent is to leave it in mostly script form.

thomaslepoix, 2020-07-03:
Haha! it is on my TODO list from the beginning because, as you said, it would be a really appreciable feature. But it seems to be lot of work so it probably won't be implemented a near future.

It was discussed few months ago [in the Qucs mailing list](https://sourceforge.net/p/qucs/mailman/qucs-devel/thread/o497TRvmj5jq3OH1WDlv6ScLs9MVgBEBfGY6zhHfjYYZGt1vxxRtHqByPfuOYx830IoaZceyHc07Hz6CcniHfelkqQyibaQR5zxXBtlm6Ec%3D%40protonmail.ch/#msg36768803).
[Here](https://github.com/Qucs/qucs/pull/371#issuecomment-141751350) is a key to deal with the Qucs equation solver, I didn't really dig it yet.

-----

Thank you!

I really appreciate simulation / measure confrontations and corner case usage repports : it always makes the software grow in quality.

My long term goal is to create a fully complete design / simulation / production ecosystem for electronics, RF, etc. Not by doing everything myself from scratch, but by making existing softwares (Qucs, KiCad, FreeCAD, OpenEMS) compatible with each other, easing their use and avoiding gaps in workflows.

Qucs-RFlayout takes place in this context.

I share this vision with few other people so we created the [Open-RFlab project](https://wiki.rfporn.org/doku.php/wiki:open_rflab). It is an early emerging project but we got a rather coherent roadmap and if you are interested in contributing, you are totally welcome! :)

(We are mostly present on a Discord)

dom11990, 2020-07-03:
Yes I share this vision myself. I would be very interested in dropping in and seeing how I can contribute. Can you send me the discord info? I was not able to find much on the website.

dom11990, 2020-07-03:
Should we leave this issue open as it is on your todo list?

thomaslepoix, 2020-07-05:
Of course, here it is! main language is french but english is welcome :)
https://discord.gg/P82fEmE

Yes good idea, so people could find this discussion and either get interested in extracting the Qucs equation solver or getting involved in the Open-RFlab project.

thomaslepoix, 2020-07-18:
OH! I just found the solution.

After running a simulation, the solved variables go in the `.dat` file with other results. For example a `x` and a `y` variables produce this :

```xml
<indep y 1>
  +1.00000000000000002082e-03
</indep>
<indep x 1>
  +9.32000000000000102668e-04
</indep>
```

So when the `.sch` parser find a variable, it's about searching for it in the associated `.dat` file and pick the value.

I close the issue and I will implement it ASAP.

closed 2020-07-18 by thomaslepoix

dom11990, 2020-07-19:
Cool, let me know when it is in! I'd be glad to try it out.

commit f003dafc (2020-08-07): fix #11 : variable support & fix #12 : -Wunknown-escape-sequence & #14 stopgap & some bugfixes