thomaslepoix/Qucs-RFlayout #9: Vias implementation

thomaslepoix, 2020-07-03:
I implemented vias as centered on the adjacent component edge rather than edge to edge but I am not sure this is right.

[image]
[image]

I would need the opinion of regulars of RF design. Thank you in advance!

dom11990, 2020-07-03:
Yeah this is rather tricky. From a designer perspective, I would prefer that the via is centered on the edge of the microstrip. However, I think the qucs model puts it on the end. So if you optimize the circuit prior to doing the oems simulation you will have deviations. I will post a question to them to see if they can answer what their model creates.

thomaslepoix, 2020-10-11:
So the answer came from the version 3 of QucsStudio from whom comes the MVIA model :

[image]

closed 2020-10-11 by thomaslepoix