thomaslepoix/Qucs-RFlayout #12: Unknown escape sequence

dom11990, 2020-08-04:
I noticed when compiling that there are some warning about unknown escape sequences. Is this expected? Perhaps you have to make "\.net" -> "\\.net" etc?

```
parser.cpp: In function ‘int parser(std::vector<std::shared_ptr<Element> >&, const string&)’:
parser.cpp:104:37: warning: unknown escape sequence: '\.'
   n_net=regex_replace(n_sch, r_sch, "\.net");

layoutwriter.cpp: In function ‘int layoutwriter(const std::vector<std::shared_ptr<Element> >&, long double* const&, const string&, const string&, const string&)’:
layoutwriter.cpp:24:14: warning: unknown escape sequence: '\.'
  regex r_sch("\.sch$");
              ^~~~~~~~
layoutwriter.cpp:25:19: warning: unknown escape sequence: '\/'
  regex r_basename("^.*?([^\/]*)\.sch$");
                   ^~~~~~~~~~~~~~~~~~~~
layoutwriter.cpp:25:19: warning: unknown escape sequence: '\.'
```

cheers
dom

thomaslepoix, 2020-08-04:
Hey!

This warning has been there for a while, it is not really problematic. I added pragmas to disable it but for some reason it is one of the few warnings you cannot disable on GCC (disabled on clang).

However I just found out how to fix it : `\.net` -> `\\.net`.

closed 2020-08-04 by thomaslepoix

dom11990, 2020-09-12:
I'm not sure if it doesn't have an effect.. it is solved by simply properly escaping the \ character with \\. Maybe Apply the fix just to be on the safe side? Technically it is invalid syntax.

commit f003dafc (2020-08-07): fix #11 : variable support & fix #12 : -Wunknown-escape-sequence & #14 stopgap & some bugfixes