thomaslepoix/Qucs-RFlayout #13: Documentation: also need freeglut3 to compile on linux

dom11990, 2020-08-04:
Title says it all...

thomaslepoix, 2020-08-04:
Which distro / version do you use?

It seems strange to me, because on [travis (Ubuntu Bionic)](https://github.com/thomaslepoix/Qucs-RFlayout/blob/master/.travis.yml) and [docker (Debian Buster-slim)](https://github.com/thomaslepoix/Qucs-RFlayout/blob/master/pack/Dockerfile.unix), `qt5-default` with its dependencies and recommended packages (`libqt5opengl5-dev`) is enough to build.

dom11990, 2020-08-04:
I'm on Debian 10

thomaslepoix, 2020-08-04:
What was the error message?

thomaslepoix, 2020-09-03:
I just installed the `.deb` package on a fresh Ubuntu 20.04 and everything works well without `freeglut3`. About OpenGL, `libqt5opengl5` is enough at runtime and thus `libqt5opengl5-dev` at build time.

closed 2020-09-03 by thomaslepoix

dom11990, 2020-09-12:
I'm running on the latest debian 10 (buster) and a new VM I just set up needed to install lglut to compile. Regardless of if a standard distro like Ubuntu comes with the package or not, I think it makes sense to include it in the dependencies list.