thomaslepoix/Qucs-RFlayout #14: Crash when attempting to read a QucsStudio schematic + netlist

dom11990, 2020-08-04:
There was a new release of QucsStudio and I'm guessing somethign in the schematic file changed. I get stoi errors when attempting to parse my files. I've attached them here for you. I was looking at the source, maybe it would make sense to subclass the parser for the different schematic files? qucs, QucsStudio 2.x, 3.x etc? Otherwise your regexes will become unreadable / unmaintainable. 

[example.zip](https://github.com/thomaslepoix/Qucs-RFlayout/files/5022688/example.zip)

cheers
dom

dom11990, 2020-08-04:
As per the parsing, obviously it works fine with regexes, however, I've found better success by having a simple space delimited parser followed by map with string keyed indecies to the individual entries you want to extra. For example, say that you have a vector of strings with all the individual values of a line, then you can go lines.at(index_map["label"]) and it will give you the label string. This makes supporting different version very easy because you only need to adjust the index_map (assuming the general structure of the line stays the same as far as the delimiter goes). It also makes it extremely readable for others.

thomaslepoix, 2020-08-05:
> Besides many new features (e.g. electromagnetic field simulator), many historical flaws were fixed. For example a single m as unit has not the meaning of milli anymore, i.e. now 3m = 3 and 3mm = 3e-3. Future versions will not be able to load old schematic files. Therefore, it's recommended to open all existing schematic and data display files with version 3.3.2 and save them afterwards.

Yes the schematic format changed. It is similar enough to seem valid to qucsrflayout but different enough to break the parser once checks passed.

Old format :

```xml
<QucsStudio Schematic 2.5.7>
<Properties>
  <View=0,0,1002,800,1,0,0>
  <Grid=10,10,1>
  <DataSet=mstub.dat>
  <DataDisplay=mstub.dpl>
  <OpenDisplay=1>
  <showFrame=0>
  <FrameText0=Title>
  <FrameText1=Drawn By:>
  <FrameText2=Date:>
  <FrameText3=Revision:>
</Properties>
<Symbol>
</Symbol>
<Components>
  <MSTUB MS1 1 240 130 26 -38 0 0 "Subst1" 1 "1 mm" 1 "10 mm" 1 "60" 1 "Hammerstad" 0 "Kirschning" 0>
</Components>
```

New format :

```xml
<QucsStudio Schematic 3.3.2>
<Properties>
View=-69,-150,1928,1226,0.906947,0,0
Grid=10,10,1
DataSet=ltc5564_match_sch.dat
DataDisplay=*.dpl
OpenDisplay=0
showFrame=0
FrameText0=Title
FrameText1=Drawn By:
FrameText2=Date:
FrameText3=Revision:
</Properties>
<Symbol>
</Symbol>
<Components>
MSTUB MS3 0 340 80 -131 -41 0 0 "Subst1" 1 "0.34e-3" 1 "0.832762mm" 1 "60" 1
</Components>
```

From what I see here, updating the internal QucsStudio -> Qucs converter could be quite easy. I'll take a look.

---

Some regexes are really useful to parse components value fields (value / scientific suffix / engineer suffix / variable I'm currently implementing). Simpler regexes are regexes only not to multiply parsing methods. But that's true a readable solution would be nice.

---

A word about QucsStudio support. The `-n` argument to specify a netlist is a stopgap as it is annoying and error prone to manually generate and update netlist each time you update the schematic. For now the only solution to avoid this is to have Qucs installed too, so that Qucs-RFlayout can convert the QucsStudio schematic to a Qucs schematic and then call Qucs to export a netlist.

So there is no QucsStudio schematic parser, only a Qucs schematic parser and a QucsStudio -> Qucs converter. With the 2.x QS versions, the conversion was simply about replacing the header and `MSTUB` components by some `MRSTUB` (and translating those by 10 to a direction).

But it will become problematic if the QucsStudio team breaks the compatibility between QucsStudio and Qucs but does not implement features like a CLI to export netlist that would be mandatory to avoid this whole hack and thus the need of a Qucs / QucsStudio compatibility.

closed 2022-08-29 by thomaslepoix