thomaslepoix/Qucs-RFlayout #19: Support defining the EndCriteria instead of max time step

dom11990, 2020-09-13:
Setting the end criteria is typically a much more time efficient way of simulating an RF structure as it simulates until a certain amount of "left over" energy is still in the system. When this is set as sufficiently small, the final s parameters will be unchanged or within tolerable error. A reasonable default value could be -1e-3. 

InitFDTD('EndCriteria',1e-3)

thomaslepoix, 2020-09-17:
As we talked privately, the default end criteria is -50dB, the max timestep is used to put a limit for non converging or converging too slowly simulations but endcriteria is (should be) anyway the normal end. but you're right it would be nice to be able to configure it, will be added.

About default value, Thorsten deprecates <1e-4 end criterias. I just tried different values on the example low pass filter, 1e-4 is acceptable while 1e-3 is frankly wrong.

dom11990, 2020-09-17:
I agree that for the general case 1e-4 is probably better. It is interesting that you say it is "wrong". How wrong are we talking here? I mentioned my differences are negligible. I think it is clear that in order to save 4x simulation time when tuning the center frequency of the filter, one would prefer to use 1e-3.

[image]

thomaslepoix, 2020-09-17:
Ah yes, negligible!

[image]

Actually, put on a comparison board, it is no so wronger. But times here are like `-50dB : 7min`,  `-40dB : 4min`,  `-30dB : 3min` so I don't think it worth it in this case.

I also mention that for -50dB, MUR or PML boundary condition make a difference, both in results and time simulation (MUR goes slower from ~ -45dB and can up to more than an hour, here it is aborted at like -47dB, ~30min), but for -40dB and -30dB, results and times are similar, here are the MUR ones.

commit fed5f6bd (2020-09-17): fix #19 : add --oems-end-criteria & refacto/bugfix : use copy ctor to reset Data & rm MainWindow Data duplicated vars
closed 2020-09-17 by thomaslepoix