thomaslepoix/Qucs-RFlayout #22: disable silkscreen value being displayed in kicad export

dom11990, 2020-09-16:
I noticed that the silkscreen values of the individual MLINs are visible in the default kicad export. I think it would be better if these had Visible: False by default. Normally the filter is expected to be a complete construct and not a coalition of individual elements. Debugging after the board is printed is difficult / impossible anyway depending on the frequency.

thomaslepoix, 2020-09-24:
Is this really relevant since it is just one click on the side bar to enable / disable it?

Also if you want your circuit as one element, you may prefer exporting to a footprint (`.kicad_mod`).

I don't understand why this would make debug difficult.

dom11990, 2020-09-24:
The silkscreen doesnt make it difficult, it is inherently difficult since it is a printed structure. I was just thinking that normally silkscreen is used to aid in debugging by providing part references or other text. In this case it wouldn't help though. I will try the other export type, maybe I mixed them up, I thought this was the footprint.

thomaslepoix, 2020-09-24:
Oh I didn't get it! disable microstrip components silkscreen and not the whole board silkscreen. You are right these texts are useless once the board is printed. Good idea.

However, if you are designing a board on kicad, I would recommend you to export your filter as a footprint and to create a schematic component "filter" to associate with.

commit 2c66e050 (2020-09-24): fix #22 : kicad_pcb : hide reference and value
closed 2020-09-24 by thomaslepoix