thomaslepoix/Qucs-RFlayout #23: move --oems-metalres-div and other to octave script

dom11990, 2020-09-22:
I think it makes more sense to move these modifiers to the octave script since this is where you try to resolve meshing issues. Currently, if i made any changes to the script and want to update this spacing you have to re-run qucsrflayout and add your changes into the resulting file. Or you edit the octave file directly which I think also shows that this should be a parameter of the octave script and not the export.

thomaslepoix, 2020-09-24:
You're right, it could be in the script land but I did that voluntarly. My point in this design was the following :

One may restart a nf2ff calcul on different frequencies or restart a simulation with different boundary conditions. Results will vary but they will be just as right. unlike to that, a wrong mesh or a wrong resolution leads to wrong results. So chosing the mesh resolution should not be a runtime option (in fact with an automatic mesher, it wouldn't even be accessible to the user).

For now meshing requires the user intervention but it should stay part of the conversion process. once a good mesh is found, It should not be modified.

The consequency of this risk of wrongness is that I totally don't want users to be able to forget the resolution they used to simulate a circuit some weeks ago. Or if one shares a simulation script, others must not have to rework the mesh to get same results as the script sharer.

So the best solution I found is to hardcode the resolution in the script. While converting and working on the mesh, it is still easy to change the resolution using the Qucs-RFlayout GUI/CLI, and then it is still possible for a user to modify it directly in the script and above of all to know which is the current resolution.

dom11990, 2020-09-24:
I suppose that makes sense, in the event that the original mesh works well. If it doesn'tt the user can just change that variable directly in the script.

closed 2020-09-24 by dom11990