thomaslepoix/Qucs-RFlayout #24: --oems-substres-div also changes air mesh?

dom11990, 2020-09-22:
I was under the impression that this should only mesh the substrate material but it also appears to affect the air cell size. I think it would be worth separating these two.

thomaslepoix, 2020-10-12:
Yes substrate resolution is actually used to mesh all non metallic materials, I is the case in most OpenEMS tutorials but you're right it may worth it so speed up simulation when there is no far field calculation.

As an other option, I thought about a script flag to reduce the simulation box (and so disable NF2FF).

What is your point about that?

dom11990, 2020-10-12:
Yeah I think a scaling of the simulation box would be a good addition.

dom11990, 2020-10-12:
The reason I opened this issue was because I was simulating a very thin substrate (200um) and because you want to have several mesh lines in there, you can imagine how long the simulation would take when the entire simulation box got that same resolution. Really the critical item was the mesh in the height dimension of the substrate. Perhaps we need to make this finer or configurable explicitly?

thomaslepoix, 2020-10-12:
Oh okay! But there is no need to care about the Z axis mesh, it is always meshed with 1 line in the metals and 2 in the substrate.

This way with metal boxes :
[image]

And this way with conducting sheets :
[image]

dom11990, 2020-10-13:
Is that sufficient? I had pretty significant differences in my simulation when I increased the density in Z..

thomaslepoix, 2020-10-20:
I don't know, trying with like 2 or 6, results were slightly different but difficult to say wich one was better. Thorsten usually uses 4 or 5 lines in the examples.