thomaslepoix/Qucs-RFlayout #25: microstrip taper support

dom11990, 2020-10-05:
Qucs 20 has a taper line, will this also be supported?

thomaslepoix, 2020-10-05:
Is it this one?

[image]

If it is, it cannot. It looks like a microstrip but it is not, both extremities are impedances instead of widths and it is independant of substrate component.

dom11990, 2020-10-05:
A taper is definitely a microstrip component, though I'd have to look at the source to see what they did. It should not be independent of substrate. Perhaps they don't actually have a microstrip model for it, only an "ideal" one?

thomaslepoix, 2020-10-05:
Yes that's it, an ideal microstrip component.. What a strange thing

closed 2022-08-29 by thomaslepoix