thomaslepoix/Qucs-RFlayout #28: Also plot group delay

dom11990, 2020-10-13:
For many (all?) applications, group delay is very important. I think it would make sense to also plot this after the simulation over frequency.

thomaslepoix, 2020-10-13:
Yes! I was waiting for a request as I am not sure how to calculate it from [OpenEMS datas](https://openems.de/index.php/CalcPort.html).. I guess you have an idea :)

dom11990, 2020-10-14:
I do.. easiest would be to use the s-parameters you already compute. Take the s21 and calculate -dphi/dw where phi is in radians and w is radians/sec.

thomaslepoix, 2020-10-15:
Ok! so I not figured yet how to perform derivation with Octave but :

- Here are phases and a phase change :

```matlab
figure;
hold on;
grid on;

phi11 = 180/pi * angle(s{1, 1});
phi21 = 180/pi * angle(s{2, 1});
plot(freq/funit, phi11, 'm-', 'Linewidth', 2);
plot(freq/funit, phi21, 'g-', 'Linewidth', 2);
plot(freq/funit, mod((phi21 - phi11), 180), 'c-', 'Linewidth', 2);

title('Phases');
xlabel(['Frequency (', funit_name, ')']);
ylabel('Phase (deg)');
ylim([-180, 180]);
set(gca, 'ytick', -180:30:180);
drawnow;
```

[image]

- Here is a phase delay :

```matlab
figure;
hold on;
grid on;

plot(freq/funit, -angle(s{2, 1})./(freq/funit*2*pi), 'g-', 'Linewidth', 2);

title('Phase delay');
xlabel(['Frequency (', funit_name, ')']);
ylabel('Phase delay (sec)');
drawnow;
```

[image]

Does that seems you correct? I am wondering about the phase delay Y axis unit (it shouldn't be divided by `funit`, isn't it?) and if the phase change should be modulo 180..

dom11990, 2020-10-15:
The phase needs to be unwrapped first, that will remove the spikes. Also, phase delay and group delay are two different quantities. Both are useful and perhaps you can plot both, but they are computed slightly differently. This will help with the implementation I think:
https://www.dsprelated.com/freebooks/filters/Phase_Group_Delay.html

thomaslepoix, 2020-10-29:
Thank you for this explaination.

- what does a RF engineer prefer, wrapped or unwrapped phases, or both?

```matlab
figure;
hold on;
grid on;

phi11 = 180/pi * unwrap(angle(s{1, 1}));
phi21 = 180/pi * unwrap(angle(s{2, 1}));
plot(freq/funit, phi11, 'm-', 'Linewidth', 2);
plot(freq/funit, phi21, 'g-', 'Linewidth', 2);
plot(freq/funit, phi21 - phi11, 'c-', 'Linewidth', 2);

title('Phases');
xlabel(['Frequency (', funit_name, ')']);
ylabel('Phase (deg)');
drawnow;
```

[image]

- So here is the phase delay. It looks better but I am still unsure about the unit.

```matlab
figure;
hold on;
grid on;

plot(freq/funit, -unwrap(angle(s{2, 1}))./(freq/funit*2*pi), 'g-', 'Linewidth', 2);

title('Phase delay');
xlabel(['Frequency (', funit_name, ')']);
ylabel('Phase delay (sec)');
drawnow;
```

[image]

- And regarding [this implementation](http://cjs-labs.com/sitebuildercontent/sitebuilderfiles/GroupDelay.pdf) here is the group delay. also here there is a unit problem, the Qucs equation `groupdelay_S21=groupdelay(S[2,1])` applied on the OpenEMS S parameter file gives a different result.

```matlab
figure;
hold on;
grid on;

phi21 = unwrap(angle(s{2, 1}));
k = numel(phi21);
grp21(1) = (-1/360) * (((phi21(2) - phi21(1))/(freq(2) - freq(1))));
grp21(k) = (-1/360) * (((phi21(k) - phi21(k - 1))/(freq(k) - freq(k - 1))));
for n = 2:k-1
	grp21(n) = (-1/720) * (((phi21(n) - phi21(n - 1)) / (freq(n) - freq(n - 1))) ...
		+ ((phi21(n + 1) - phi21(n)) / (freq(n + 1) - freq(n))));
endfor
plot(freq/funit, grp21, 'g-', 'Linewidth', 2);

title('Group delay');
xlabel(['Frequency (', funit_name, ')']);
ylabel('Group delay (sec)');
drawnow;
```
[image]
[image]

dom11990, 2020-10-29:
The unit should be seconds. I think your mistake is the scaling, it should be divided by 2pi, not 360 since the angle from unwrap is probably in radians.

grp21(1) = (-1/**2*pi**) * (((phi21(2) - phi21(1))/(freq(2) - freq(1))));
grp21(k) = (-1/**2*pi**) * (((phi21(k) - phi21(k - 1))/(freq(k) - freq(k - 1))));

I don't really care much about the raw phase; group delay and phase delay contain all the information I normally need.

thomaslepoix, 2020-10-30:
Well seen, it's ok for group delay! Does the phase delay seems good to you?

Do you think no engineer needs raw phases?

thomaslepoix, 2020-11-01:
So I implemented phase response, phase delay and group delay for transmission parameters. Please close this issue if everything seems good to you.

commit f505a36e (2020-11-01): fix #28 : oems : add phase response & phase delay & group delay
closed 2021-02-03 by thomaslepoix