thomaslepoix/Qucs-RFlayout #34: Octave 9 issues

delta-atk, 2024-04-30:
I tried the export from qucs-s and got some issues with Octave 9. I had to change the following things in the octave code to get it to run.
* Octave 9 does not use `.+` or `.-` operators, i had to change them to `+` and `-` so e.g. the line `ReZ .+ linspace(-ImZ, ImZ, 256);` does not work and has to be `ReZ + linspace(-ImZ, ImZ, 256);` now
* `physical_constants` is not known, i had to delete this line and change the term `c0` in the `%%%% VARIABLES` section to `pkg load miscellaneous` and `physical_constant("speed of light in vacuum")`, and a few lines later `EPS0` to `physical_constant("electric constant")`
* Also i had to comment out the lines `pkg load openems;` and `pkg load csxcad;` at the beginning but there is already a comment in code to uncomment it to use Debian packages so i guess this is maybe not related to Octave 9
* Plots are unresponsive, but the generated images are ok. Also when using the Octave GUI running the script immediately crashes it, just CLI does work. These error could also be Windows issues i guess

thomaslepoix, 2024-05-02:
Hi, Thanks for reporting.

- Indeed, fixed.

- This is the normal behavior because the official installation process involving `addpath('/opt/share/openEMS/matlab');` in `octaverc` is a bit hacky, while some distros provide a properly packaged octave interface, included with `pkg load`. By default those lines are commented out so I assume you either used the `--oems-pkg` argument, either checked the `Use Octave packages` box in the GUI.

- I just observed an unresponsiveness on NixOS too, I fixed it by using the `octaveFull` package, which just seems to [embbed the Qt GUI](https://github.com/NixOS/nixpkgs/blob/nixos-unstable/pkgs/top-level/all-packages.nix#L17447-L17451) compared to the `octave` one. Now it's possible to zoom, etc. Don't know how this would translate to the Windows releases, especially since you obviously has the GUI installed. Here are some before/after screenshots, is this what you encountered?
[image]
[image]

- I could catch this message error trying to launch the script from the Octave GUI `ERROR : Unknown argument : --gui` so I assume Octave forwards its own arguments to the script. Anyway, I don't recommend launching the script from the GUI since many control features are accessed trough the CLI.

delta-atk, 2024-05-02:
I see, i checked the `Use octave packages` box, i think it confused me because i thought it is necessary to use octave in general.  

Yes this is very similar:
[image]

I realized now that octave-cli cannot use qt but only octave.exe. I built the current version using arch wsl and used it to generate a new .m. Now it runs out of the box when using octave on Windows, and the plots are fine from octave.exe. For me, there are no more errors now, thank you.

closed 2024-05-03 by thomaslepoix