thomaslepoix/Qucs-RFlayout #40: Build Fail Ubuntu 22.04.5 LTS

fossy411, 2025-06-02:
Beautiful project please don't let it fail, if I can help in any way let me know.

ISSUE: will not build in Ubuntu 22.04.5 LTS
(Debian 12 build and binary works perfectly)

PROBLEM: Compatibility issue with C++23 compiler

SOLUTION: tested and is working

ADD:  compiler directives

#ifndef unreachable
#if defined(\_\_GNUG\_\_)
#define unreachable() __builtin_unreachable()
#else
#define unreachable() ((void)0)
#endif
#endif

to files:

pac.hpp
subst.hpp
mcorn.hpp
mcoupled.hpp
cross.hpp
mgap.hpp
mlin.hpp
mmbend.hpp
mopen.hpp
mrstub.hpp
mstep.hpp
mtee.hpp
mvia.hpp

ALTERNATIVE: possible suggestion 

NOTE: THIS HAS NOT BEEN TESTED!

Define inline helper function

#include \<cstdlib\>  // or another header if needed

[[noreturn]] inline void unreachable() {
    __builtin_unreachable();
}

thomaslepoix, 2025-06-04:
Duplicate of #36

Hi, thank you !

As far as I remember `std::unreachable`, which is easily backportable, is the only C++23 feature currently used. But more generally, I would rather recommend users to use an up to date compiler than providing support for old C++ standards.

According to [this table](https://en.cppreference.com/w/cpp/compiler_support.html#C.2B.2B23_library_features), Clang15 which is [available on Ubuntu 22.04](https://packages.ubuntu.com/jammy/clang-15) should be enough. Alternatively, [LLVM repository](https://apt.llvm.org/) provides cutting edge Clang versions for many distros. Otherwise, I'd suggest [Nix](https://nixos.org/) to avoid dependency issues on more exotic/outdated distros.

fossy411, 2025-06-05:
I tried Clang-15 and Clang-16 from both the built-in Ubuntu package manager repository and LLVM. Both would not build.

However, I installed the g++-12 included in the Ubuntu package repository and it built perfectly. (Ubuntu defaults to g++ version 11.4.0 for some reason)

I had to manually add `sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-12 30` , but this version worked flawlessly. 

Thank you....nice software keep up the good work.

closed 2025-06-05 by thomaslepoix